SynthstromAudible/DelugeFirmware #2710: Automations recorded/entered dissapear on song reload.  (vol & pan; modFX rate; reverb depth; master pitch; decimation; bitcrush)

percussijan, 2024-09-24:
### Please describe the problem:

Above parameter automations do not save with the song. After song reload all recorded or manually entered automations are gone except delay time/rate and filter cutoff/resonance. 

### What is the expected behavior?

All automations save with the song and are still there after song reload.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.2.0-NIGHTLY-5264A54

### If possible provide the steps to reproduce the issue and upload additional media:

Enter or record automations in song or clip view. Save song. Reload song.

m-m-adams, 2024-10-02:
Confirmed - seems to be generally failing to save anything other than the base parameter values. Modifying these parameters between clips is also not saving

m-m-adams, 2024-10-02:
Actually I think they might be entirely unsaved

PR #2798 (2024-10-09): Bugfix - leave space for params saved as attributes
Fix #2710 

The kit param xml tag was being closed on write, so the params expecting to be read as attributes get skipped looking for the next "<"
closed 2024-10-09 by seangoodvibes