SynthstromAudible/DelugeFirmware #2756: Changing the type of Audio Clip (Sampler, Looper, Player) is not visually indicated on the 7SEG display

baymud, 2024-10-04:
### Please describe the problem:

In order to change the type of audio clip to sampler or looper (to engage input monitoring) you have to turn the select knob, but there is no visual indication of this selection being done on 7SEG (it just says the name of the track, in this case "AUDIO 1")

### What is the expected behavior?

When the select knob is turned it should display the audio clip type ("SAMP", "LOOP", "PLAY" maybe?)

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

baymud, 2025-01-08:
I wouldn't say this is completed yet, as #3234 only fixes the display in the clip itself, not in song view, where you can also change the clip type, but still can't see that reflected on the 7SEG display (@m-m-adams mentioned he was going to look into that)

m-m-adams, 2025-01-08:
It fixes both already, I updated it

PR #3234 (2025-01-08): show a popup for audio clip type change on 7seg
Fix #2756
closed 2025-01-08 by m-m-adams