SynthstromAudible/DelugeFirmware #2758: Mutable Reverb model is enabled regardless of which model was saved in a given song

ok-reza, 2024-10-05:
### Please describe the problem:

Was playing old songs and they didn't sound the same. Looks like they all used the FREEVERB reverb originally, but MUTABLE is the model enabled.

### What is the expected behavior?

Have the reverb model saved to the song switch to that reverb. Seems like Mutable is default universally currently.

### Is there a relevant Pull request?

possibly? #2404

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2

m-m-adams, 2024-10-05:
I can't reproduce, saving and loading songs with freeverb is working fine. Can you share the song please? (just the xml is fine)

m-m-adams, 2024-10-05:
Occurs on songs from official 4.1.4 and community 1.0, the check eliminated too many versions

PR #2776 (2024-10-05): Bugfix/UI niggles
Fix #2769 
Fix #2760 
Fix #2758
PR #2777 (2024-10-05): Bugfix/UI niggles
Fix #2769 
Fix #2760 
Fix #2758
closed 2024-10-05 by seangoodvibes