SynthstromAudible/DelugeFirmware #2759: Hitting BACK a couple times from the main menu while 'SAVING SETTING' is displayed will cause a crash.

ok-reza, 2024-10-05:
### Please describe the problem:

Entering the main menu and then exiting with repeated BACK presses will cause a crash. Sometimes it only takes one BACK while 'SAVING SETTING' appears to trigger the crash, but it's more consistent if you press multiple times.

Additionally, once I reboot the Deluge, I get a Start Up Song error, despite having no start up song previously enabled. it displays "START UP OFF,  REASON:'

### What is the expected behavior?

No crash on exit.
No issue with start up song displays.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2 beta

m-m-adams, 2024-10-05:
I can't reproduce this - I'm slightly suspicious that your SD card might have a problem based on the symptoms, maybe try repairing it on a computer?

ok-reza, 2024-10-05:
im suspecting it might be an issue of the slower read/write speed then? im using the sd card that the deluge comes with. but i'll check w repairing too!

m-m-adams, 2024-10-05:
That message should only be present if you have a song named SONGS/__STARTUP_OFF_CHECK_ on your card - do you see one? 

It would have been created by the startup song feature failing at some point in the past

pmatila, 2024-12-06:
I'm seeing this kind of thing a lot on 1.2, so much that it's been keeping me on 1.1. For me a fairly reliable reproducer is settings -> midi -> devices -> upstream port 2 -> mpe -> in -> lower zone -> change value (and back, if you like), and then hit back quickly until it it has exited the menu and saved (as in, just keep hitting back without waiting for feedback):
[image]

I'm also using the SD card that came with Deluge. There *was* the _STARTUP_OFF_CHECK file present on today's first crash, but after removing that file on PC and running fsck on the card filesystem I can still reproduce the crash pretty much at will, tried several times, rebooting each time.  ... except that after testing it on some random CV related setting from which I slowly backed out and didn't crash, I seem to have lost the ability to reproduce it anymore.

I remember seeing something fishy wrt settings when moving between 1.1 and 1.2 when I tried it earlier, maybe it's just related to the version change, some difference in the settings. I don't remember seeing it crash on 1.1 ever before when backing out of the menu, but now if I go back from 1.2 to 1.1 I can reproduce it *there* just as easily as it was crashing on 1.2 before. And going back to 1.2, it's still okay, can't make it crash :zany_face:

ok-reza, 2025-02-10:
this still can occur for me on latest nightly 1.3 (2/9)
both me and pmatila received a E258 error. my most recent one was a E259 error.

seangoodvibes, 2025-11-22:
Same issue reported on discord today:

[image]