SynthstromAudible/DelugeFirmware #2764: Key auto resets to C in Keyboard Mode when pressing SCALE twice

uke4peace, 2024-10-05:
### Please describe the problem:

Aloha! New to Deluge. Just got a used one this week and am doing a deep dive to test it works properly. Excuse me if the following is a feature and not a bug.

Key auto resets to C in Keyboard Mode when pressing SCALE twice.

To reproduce:

Synth mode > keyboard mode > hold SCALE + rotate SELECT knob to choose any key other than C >  press SCALE > press SCALE again

Video: [https://drive.google.com/file/d/1EB0C6CI7AVKZ0pmeP_jxGPqjXMP8rh6c/view?usp=drivesdk](url)

Mahalo.

### What is the expected behavior?

Pressing SCALE 2x in keyboard mode to double check what scale mode is currently active and returning back to keyboard mode in the key that was chosen.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

1.1.1-DEV-9FBD773A

### If possible provide the steps to reproduce the issue and upload additional media:

To reproduce:

Synth mode > keyboard mode > hold SCALE + rotate SELECT knob to choose any key other than C >  press SCALE > press SCALE again

Video: [https://drive.google.com/file/d/1EB0C6CI7AVKZ0pmeP_jxGPqjXMP8rh6c/view?usp=drivesdk](url)

m-m-adams, 2024-10-05:
Hey yeah what's happening is the first press clears the song's scale, the second one deluge tries to grab the most likely scale from the notes you've entered. If there's Cs and As in it but nothing outside of a C major scale (so no flats or sharps) it'll choose C major

m-m-adams, 2024-10-05:
It's fixed in the nightly anyway, the first press of scale now shows the current scale instead of clearing it

closed 2024-10-05 by m-m-adams

uke4peace, 2024-10-06:
Thank you Mark and dev team for the update and the work you do. Btw, there are some stock kits that are missing sounds? Like the Chuck Bundick kit is missing sounds for Kic1 & Kic2, pads 5 & 6 from the top. There was another kit missing one too. Is this something your team handles?*Emailing from mobile. Excuse typos.--Best,Erik w/ Oléha🏝 Vacation music for the soul💛 Classy, feel-good entertainment for all!714.865.5027Music Links: linktr.ee/oleha.musicinstagram.com/oleha.music

uke4peace, 2024-11-27:
Hello Mark, before you said this issue was fixed in the nightly where the
pressing scale causes the scale to be reset to C. Is the nightly version
the one that keeps updating before an official open source community
update? Thank you.
*--*
*Best Regards,*

*Erik w/ Oléha*
🏝 *Vacation Music for the Soul*
*💛 Feel-good lounge entertainment for all!*
*714.865.5027*
*Instagram @olehamusic <https://www.instagram.com/olehamusic>*
*Electronic Press Kit:  Desktop
<https://drive.google.com/file/d/1YWV65ANaO3BVfkZ2FNvIDffC0GGRnjyO/view?usp=share_link>
 |  Mobile
<https://drive.google.com/file/d/1YfOV_xiLl8vWVQb5sgOzxokhYsvHNefe/view?usp=share_link>*
*Live Performance Video:  One-Man Band
<https://youtube.com/playlist?list=PLiNiV3OtyB_VpyZXMhTwekJ9oe2NG5OPj&si=pHbhq4tbMwCBp0PS>
 |  Duo
<https://youtube.com/playlist?list=PLiNiV3OtyB_WSZNm6HiydLdCT7Rd6Pzp-&si=4p_UKi1c1rKabENC>*