SynthstromAudible/DelugeFirmware #2767: Some OLED menus do not keep the highlighted section in the 2nd row when scrolling

ok-reza, 2024-10-05:
### Please describe the problem:

A newer feature was implemented for OLED menus where, when scrolling down menu options, the highlighted section would remain the 2nd row. The highlighted section only reaches the bottom third row when it reaches the end of that menu's list.

Currently, any SONG or PRESET browser and SAMPLE browser does not respect this change.

### What is the expected behavior?

To be consistent, while scrolling down/up menu items, the highlighted section should remain as the 2nd/middle row for the SONG, SYNTH, KIT, MIDI, and SAMPLE browser until it reaches the first or last item in the menu.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2-beta

m-m-adams, 2024-10-05:
@nikodemus any ideas?

nikodemus, 2024-12-08:
I'll take a look in next week, but I'm guessing duplicated logic somewhere.

nikodemus, 2024-12-08:
Yeah, the change I did was for things driven from SoundEditor stuff (I think, I don't actually remember), this seems to be driven from ContextMenu.

nikodemus, 2025-01-02:
So, both context menus and browsers do this.

Fixing the browser reliably looks hairy: I think that might better wait for browser refactoring or replacement -- the logic there isn't easy to follow.

Context menus only have two rows atm, but there the highlighted row should IMO be the top row when scrolling down, and the bottom row when scrolling up -- so that you always see the next item in the direction you're heading.

Context menus could also use a visual indicator that there's more up/down.

Metamere, 2025-08-05:
I didn't implement this, but it should be a lot easier to do so if my PR #3994 is merged.