SynthstromAudible/DelugeFirmware #2816: Crash

Evilpilli, 2024-10-16:
### Please describe the problem:

Crashes when selecting audio as oscillator type, and changing the filter in automation view. 

### What is the expected behavior?

For it not to crash. 

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

m-m-adams, 2024-11-06:
I can't reproduce this - closing for now, happy to try again with more details

closed 2024-11-06 by m-m-adams

Evilpilli, 2024-12-07:
> I can't reproduce this - closing for now, happy to try again with more details

Tested this again on the recently nightly (Audio as input as oscillator is not a feature I use allot) and I can no longer reproduce the crash either. So seemingly fixed.