SynthstromAudible/DelugeFirmware #2819: Turning the <> encoder one notch to display current bar info shows an incorrect value

soymonitus, 2024-10-17:
### Please describe the problem:

On a brand new song when in clip view, if you turn <> once, it prints "1 : 1 : 1" but if you hold shift and turn <> it starts in "1 : 0 : 0" and goes to "1 : 0 : 1", "1 : 0 : 2" and so on. Should it display "1 : 0 : 0" when you turn it without shift to know the current pattern length?

### What is the expected behavior?

It shows correct pattern length ("1 : 0 : 0")

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

seangoodvibes, 2024-10-17:
I think one relates to the current scroll position while the other is the length?

Try what happens when you turn <> on another page

soymonitus, 2024-10-25:
If that is the case, then it feels weird to have something mean different things. That is, the 3 numbers mean different things depending on the context?
Shift + <> :    1 : 0 : 0 -> this is just length and subdivisions of the length?
Press <> :     1 : 1 : 1  -> Are you saying that here each number means a different thing here?

ok-reza, 2024-10-29:
hm, i think you're right @soymonitus 
it should never be possible to get a 0 after the first bar number. eg, the first beat should read 1:1:1 not 1:0:0.

see ableton's implementation and x:x:x values for reference:
[image]

soymonitus, 2024-10-29:
@ok-reza I tested and now understand what it's all about. when moving one notch the <> encoder, you see the current bar in focus (in reference to the most left cursor position, say you have a 4 bar clip, when moving <> it will print 1:1:1 on first bar, 2:1:1 when you move to the second bar and so on till 4:1:1
When using Shift + <> you are adjusting clip length, so what the screen prints is not cursor or screen position but clip length in bars, subbars, etc. so in the previous example, no matter which bar you are in focus, if you press shift + turn it will print 4:0:0 as the clip is 4 bars long, and as you increase pad by pad it goes 4:0:1, 4:0:2, etc.

I think the behavior is expected but a bit confusing if you don't read any explanation

ok-reza, 2024-10-30:
ah, you are totally right! makes sense and should be considered expected.

issue should get closed but i think the following should be kept in the back of our minds:

1) in future documentation, making this distinction clear would be helpful
2) on OLED, perhaps a "Clip/Screen Position" and "Clip Length" title in the pop up would be useful. in my opinion, the pop up covers most of the screen anyway, so perhaps it's own custom screen that takes the whole display up would be valuable (benefits being: aforementioned title context and larger number values displayed which will be easier to read from a distance ala 7SEG).

RJ-Eckie, 2024-11-01:
> 2\. on OLED, perhaps a "Clip/Screen Position" and "Clip Length" title in the pop up would be useful. in my opinion, the pop up covers most of the screen anyway, so perhaps it's own custom screen [...]

When adapting the UI, I would prefer something that is less intrusive rather than more. With the instrument title being centered there's often some space on the left and right sides. Possibly putting **clip position** bottom left and **clip length** bottom right could be enough visual language to signal that these values are indicating different things?

closed 2024-12-01 by m-m-adams