SynthstromAudible/DelugeFirmware #2830: extending note duration creates unwanted extra note-on/off events over midi

duncanrmi, 2024-10-20:
### Please describe the problem:

midi clip, 8th note resolution, enter a single note & extend it by holding it & clicking on following pads (i.e. one pad fully lit, one or more successive pads dimly lit).
over midi, this produces a separate note-on & note-off for each of the pads, not one note-on at the start of the grouping & one note-off at the end.

### What is the expected behavior?

one note-on, one note-off after a duration set by the number of lit pads.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

chopin, but have seen it with beethoven too.

### If possible provide the steps to reproduce the issue and upload additional media:

[image]

seangoodvibes, 2024-11-03:
I don't think this is a bug

I think this is the midi monitor seeing messages come through on all the deluge midi ports

The midi monitor / any device interacting with the deluge should be able to select a single port to see the actual of messages being sent per port

duncanrmi, 2024-11-03:
yeah, but I discovered it while using the deluge to control a NDLR* over standard DIN midi. it also happens when I hook the deluge up to other synths over midi or cv/gate. my minimoog doesn't like the multiple gates per note, & stays gated until I remove the multiple triggers. if I slow the deluge down enough, I can hear them all going past.
appreciate yr point about the midi monitoring, but I've tried this on several apps & seen the same.

(*NDLR plays a pad sound,a pedal bass & two melody arpeggiator/sequencers in a preset key; incoming notes move its output through the degrees & variations of chords & lines in that preset key. doing this with the deluge results in the pad sound being triggered four times in rapid succession, while the melody lines stumble.)

duncanrmi, 2024-11-03:
"remove the multiple triggers" by zooming in & reducing every note to the shortest possible. this is tedious, & I wish there was a gold-knob or black-knob function for note length editing, &/or a default note length in the deeper settings. maybe stored with the synth preset? anyway.... recording into a cv/gate pattern over midi doesn't seem to cause any problems, because I'm playing really short notes on a keyboard or sending them from (e.g.) a darktime or something.