SynthstromAudible/DelugeFirmware #2834: Midi clip editing when using transpose

chaverilli, 2024-10-21:
### Please describe the problem:

When editing midi clip whilst sequence is playing and also using transpose the screen jumps to the lowest notes when clip returns to zero

### What is the expected behavior?

The view should not jump to the lowest midi notes

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2024_10_18-655127d.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on
Song- Grid View
Insert Midi track
Insert transpose track
Play both tracks
Edit notes on Midi track

duncanrmi, 2024-10-22:
nuisance, isn't it? I have to keep muting my transpose track & setting it back to the first note before getting to edit any of the other patterns.  I _know_ they're transposing- you don't need to try to show me! I just want to edit the notes in the pattern.... 

I think it stops doing this if you switch transpose to chromatic, but that ruins the transposing...

chaverilli, 2025-05-16:
> @chaverilli can you confirm that this is still an issue with the 1.3 beta?
> Im pretty sure we fixed this
> Thanks!

I’ll take a look… haven’t updated beta for a few weeksSent from my iPhone