SynthstromAudible/DelugeFirmware #2849: OLED display refresh rate gets laggy depending on clips playing

ok-reza, 2024-10-29:
### Please describe the problem:

A song containing 4 instruments and 5 total clips seems to produce OLED display lag when the drum clip is playing back with other clips. If I mute the drum clip, the display lag is greatly reduced. I'm wondering if this could be helpful in finding the display lag culprit(s).

The display lag can be viewed while remaining in Song View and launching clips from there. The title of the song is long enough that it will always scroll, making it easy to keep an eye on the display's refresh rate.

Song contents:
1 Kit Clip (8 sample rows, kick has sidechain send of 50)
1 Synth Clip (1 sample played as a monophonic arp)
1 Synth Clip (1 short sample played as a one shot with delay)
1 Synth Clip (Bass patch with volume reacting to sidechain in a negative amount. Two clip instances)

All 10 samples used are 44.1khz, 16bit, and mono with the exception of one short sample that is 24bit.

[10B.Working In A Cold Room 2.zip](https://github.com/user-attachments/files/17561952/10B.Working.In.A.Cold.Room.2.zip)

### What is the expected behavior?

OLED display lag should only occur under heavy performance load.

I'm also noticing that any time a clip is launched/muted, the scroll position of the song is reset. Personally, I don't think this is necessary, but I wonder if this adds any additional 'stress' to the display's performance.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2 beta from 10.19

### If possible provide the steps to reproduce the issue and upload additional media:

Begin playback.
Launch rows 2, 3, & 4. (columns 2, 3, 4 in grid view)
Observe OLED display refresh rate while drums aren't playing.
Launch row 1 (drums) (column 1 in grid view)
Observe OLED display refresh rate lag while drums are included in playback.
You can mute all rows except the drums and witness similar lag.

m-m-adams, 2024-11-03:
I don't think this is necessarily a bug - we'd probably have to cull more voices to fix it

stellar-aria, 2024-11-22:
can we bump the frequency of UI renders/output? or will that start to cut into audio performance?

closed 2024-12-10 by stellar-aria