SynthstromAudible/DelugeFirmware #2859: Pitch bend values of a MIDI track not being saved to the song

songsta, 2024-11-01:
**Please describe the problem:**

The Pitch Bend range settings don't save to the song file or to the a MIDI instrument preset.

I have my Linnstrument and Peak plugged into my Deluge. The Linnstrument is set on MPE mode, with PB set to 24 (more than enough for me). The Peak has a max PB range of 24. I have a MIDI track set up with the MIDI track settings to convert poly to mono with MPE checked. In the PB range settings I set the poly PB to 24 and the normal PB to 24. Everything then works as expected - when I PB on the Linnstrument over 2 octaves the Peak correctly maps to this and PBs over 2 octaves. 

I then save my MIDI track as a template and save the song and turn off the Delly. When I turn it back on and load the song, the PB ranges have reverted to 2 for normal and 48 for MPE. Now the Peak and Linnstrument PB ranges are no longer synced and I then have to change these again in order to play my track properly.

What is the expected behavior?

When I turn on the Deluge and reload the song the Pitch Bend values should be what I had set them to and not the default of 2 for normal and 48 for Poly. 

Is there a relevant Pull request?

No response

What hardware did you reproduce it with?

OLED

What firmware did you reproduce it with?

Beta 1.2

What is the firmware name:

m-m-adams, 2024-11-03:
Present in 1.1, not present in 1.0

stellar-aria, 2024-11-22:
@songsta can you please test the firmware in #2951 and verify that it acts as you expect?

closed 2024-11-22 by stellar-aria

songsta, 2024-11-24:
> @songsta can you please test the firmware in #2951 and verify that it acts as you expect?

@stellar-aria Thanks. Should I check nightly or Beta?

songsta, 2024-11-24:
It's working nicely! Thanks!

PR #2951 (2024-11-22): Save bend range for MIDI instruments
Fixes #2859
PR #2953 (2024-11-22): Save bend range for MIDI instruments
Fixes #2859