SynthstromAudible/DelugeFirmware #2890: Freeze (E411) after pressing pad(3, 13)in clip instrument view (debug only)

RJ-Eckie, 2024-11-08:
### Please describe the problem:

When in instrument clip view (synth, kit, midi) and pad 3, 13 (mod fx feedback) has been pressed, device will freeze on basically any action. Any parameter change, pressing any main grid pad. Display will display Error E411 most of the time. Sometimes display is a scrambled error message, sometimes display is unchanged.

Changing preset after pressing (3, 13) but before doing anything else avoids issue.

CV clips seem to be unaffected, as are song view, arranger view, automation view. No-repro on any release build. (Source, nightly, beta.)+

### What is the expected behavior?

No freezes

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

EB804DAA

### If possible provide the steps to reproduce the issue and upload additional media:

1. Open any synth, kit or midi clip view
2. Press pad (3, 13)
3. Turn a gold knob or press a grid pad

m-m-adams, 2024-11-12:
I can't reproduce this either, not getting a crash

RJ-Eckie, 2024-11-12:
> I can't reproduce this either, not getting a crash

Alright lemme try this on different versions as well

RJ-Eckie, 2024-11-14:
Can only reproduce this in debug build of current source (7759705A)

Test 1: Existing song, sequence some random notes. Then: press (3, 13), press any random note after that
Test 2: New song, new synth, sequence practically all pads, turn some gold knobs to see whether they still work. Then: press (3, 13), turn a gold knob

https://github.com/user-attachments/assets/4cf22421-f068-4586-a98e-048a33b111fa

m-m-adams, 2024-11-14:
Wow that's absolutely bizarre! I was trying to use the shortcut with shift and not just press the pad so I guess that's why it wasn't triggering

m-m-adams, 2024-12-01:
I don't have a cause yet but the crashes come because the activeModControllableModelStack has a garbage but not zero patched param collection