SynthstromAudible/DelugeFirmware #2897: Audition pads stay on when editing non-playing clip

RJ-Eckie, 2024-11-10:
### Please describe the problem:

In song mode, when editing a clip that is not playing and using the audition pads, the pads stay lit after releasing them.

### What is the expected behavior?

When releasing an audition pad, pad should return to being unlit

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### If possible provide the steps to reproduce the issue and upload additional media:

Song View (either layout,) create a track with multiple clips
Arm one of the clips for playing
Press play
Edit a clip that is not playing
Use the audition-pads to audition sounds

m-m-adams, 2024-11-12:
I can't reproduce this - I followed the steps and the pads are correctly dimming after I release

RJ-Eckie, 2024-11-12:
> I can't reproduce this - I followed the steps and the pads are correctly dimming after I release

Ok let me try what versions this is does/doesn't occur on and I'll post a video of the results

RJ-Eckie, 2024-11-14:
It looks like this only happens in kits. I've been able to reproduce this back to release v1.1 (current downloadable build). Not in official. Issue only occurs when playing

https://github.com/user-attachments/assets/23d8b412-60d2-4ce4-a9a0-aafc2f4b29f8

m-m-adams, 2024-11-14:
Putting this as future release - we've already released firmware with this bug so I don't think it should hold up releasing 1.2 

That said if there's a fix we should bring it to 1.2 as well

wyvc, 2025-02-17:
I'm also getting this issue. To add, when the bug is present parameters are not updated to the selected kit row until playback stops (shown at the end of the video). I will copy my description to this issue:

Getting this bug on c1.2 with kit clips.
-Make 2 clips of the same kit (or copy the kit clip.
-While the first version is playing (and only while the deluge is playing - issue not present when not currently playing), enter the second clip to edit. 
-While in the non playing version, pressing audition pads causes the pad light to light up but not dim (light gets stuck on) after pressing a different audition pad. Also, the song button flashes blue (like an error indicator that I need to be in song view for some reason - this might not be related to the bug and just a feature? to indicate you're not editing the playing clip).
-Changing any parameters for the selected row does not update until after stopping playback.
-Updating the UI via scrolling etc. updates stuck audition pad lights momentarily.
-I reinstalled the .bin by pressing shift during startup and it still happens. I don't recall this behavior before so I don't know when or what started it.

https://github.com/user-attachments/assets/dd2ef11c-1dee-471f-b3e3-e0bc032397a7

Sorry for the vid quality