SynthstromAudible/DelugeFirmware #2900: Track quick-type-select not always usable

RJ-Eckie, 2024-11-11:
### Please describe the problem:

When:
- In grid layout
- Default mode: launch
- Creating a track in edit mode (blue + empty pad on empty track)
The quick-type-select menu pops up, but can't be used because both hands are already holding a pad, likely on either side of the device. When either pad is released, choice of clip type is finalized.

This means the user has to choose between either:
- Using the quick-type-select menu, but the clip won't be entered on release (one hand holds clip pad, other turns select)
- Clip will be entered on release, but type can't be quickly selected (one hand holds blue, other pushes clip pad)

### What is the expected behavior?

In default launch mode, user can create a clip in edit mode (clip will be entered after release) **and** make use of the quick-type-select menu

One possible solution would be for the quick-type-select menu to stay on screen as the blue pad is released - allowing them to turn the select knob with the right hand - and the type selection finalized when the clip pad is released.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

m-m-adams, 2024-11-12:
You don't need to hold blue to create a clip

closed 2024-11-12 by m-m-adams

RJ-Eckie, 2024-11-12:
> You don't need to hold blue to create a clip

I know and I've included that in the issue - behavior is different between holding blue and not holding blue while creating a clip