SynthstromAudible/DelugeFirmware #2908: Session macros not available as a sidebar control in drum velocity keyboard view

seangoodvibes, 2024-11-14:
### Please describe the problem:

Session macros not available as a sidebar control in drum velocity keyboard view

### What is the expected behavior?

Session macros are available in drum velocity keyboard view sidebar

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

m-m-adams, 2024-12-01:
Are any columns available?

m-m-adams, 2024-12-01:
This is a fairly large refactor - moving it to discussions since it's definitely a feature request and not a bug