SynthstromAudible/DelugeFirmware #2916: Tempo fluctuates with external clock sync

seangoodvibes, 2024-11-15:
### Please describe the problem:

Reported by two people on discord:

@soymonitus 
> I have noticed in nightly that when I have the Deluge clock synced to external clock i see no exact clock, like the host is set to 120 (Loopy Pro) and deluge shows in the screen 120.35 and then after some time, it starts to be slightly out of sync
> someone else noticed this?

@heek
> Yes, I noticed this. If I load a song with audio samples and play it with external clock sync, the tempo starts to be unstable. Maybe because of loading these audio samples to internal memory or? Because during the second playback of the same song the sync  is stable.
> 
> Stable I mean that the tempo of 120bpm for example floating between plus minus 1 bpm.
> 
> Maybe cpu limit? sometimes playbutton blinking
> 
> Beta is doing that sync issue too, song tempo floats a lot.

### What is the expected behavior?

Tempo sync is stable

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly, Beta

soymonitus, 2024-11-15:
It was happening to me with an empty project with just a kit clip

m-m-adams, 2024-11-16:
120.35 is almost exactly on, that's an error of around .1ms per tick. It's possibly just a normal amount of midi clock drift

soymonitus, 2024-11-17:
But getting a small out-of-sync time phase, after some time playing, is not normal, right?

m-m-adams, 2024-11-18:
It depends how big it was - there is a bug with loopy pro where it occasionally sends song position pointers that don't match the current position of the song and knock the deluge out of sync

stellar-aria, 2024-11-22:
Closing as wontfix due to being representative of a bug on the clock source side, not the Deluge.

closed 2024-11-22 by stellar-aria