SynthstromAudible/DelugeFirmware #2920: Tempo doubling with Ableton Live when deluge is master clock

todd-gochenour, 2024-11-15:
### Please describe the problem:

 I just encountered an issue with community c1.1 using Ableton live as a slave to deluge using a MIDI DIN closed ring network.  The deluge also has a USB connected midi keyboard attached.

The issue occurs when I press play on the deluge and playback on both devices is doubled.  Eventually ableton gives up and says 'the midi sync signal has been lost.'  After disabling deluge midi clock input this issue was resolved.

My hope was to be able to assign clock master to either device by pressing play on one so that the other would receive the start tempo and fall-in automatically.  Ableton starting deluge worked fine when both deluge settings for midi clock input and output are enabled.  Deluge starting Ableton failed until I turned off midi clock input.  

I see a similar issue resolved last year where #620 says there are multiple midi devices that can be distinctly configured as input/output enabled.  So is my USB midi keyboard a second device?  I don't see a device list in the global midi menu.

Perhaps the Deluge could be smart enough to know that when it starts the clock it should then ignore incoming clock messages.  Or perhaps the problem is with Ableton sending a separate delayed set of messages.  I activated the Ext button on Ableton Live's transport. I assume this would disable it starting a second clock.  Perhaps I'm incorrect.  I have seen many posts in Ableton forum about issues with syncing to an external clock.  

Still researching...

### What is the expected behavior?

The deluge ignores incoming clock messages when it is master.

### Is there a relevant Pull request?

#620

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### If possible provide the steps to reproduce the issue and upload additional media:

connect midi network. Configure Ableton to sync with the EXT button.  Press play on deluge. Midi clock input is enabled.

acidduckling, 2024-11-15:
Hi!

This sounds like you're doing a MIDI no-no. While MIDI is a protocol for communication between instruments, it's not the same as a computer ethernet network - you cannot have a "MIDI ring network" like a computer network. as you've described. You should not feed a MIDI clock back into a master clock device - you'll get exactly the problem you've described (clock doubling). I hope I've interpreted your issue correctly - forgive me if I've misunderstood.

For more complicated MIDI routing scenarios, there are MIDI boxes like the iConnect range of products (among others - I mention  iConnect because I personally  use an iConnectMIDI4+ for complex MIDI setups), which can do control and clock filtering. 

That being said, do you experience the same issue with the official firmware?

seangoodvibes, 2024-11-16:
Have you tried the 1.2 beta? We fixed a bunch of clock issues with the 1.2 beta so it would be worth trying!

m-m-adams, 2024-11-16:
You should see 3 midi ports for the deluge, only select one for the clock source in ableton. If you don't and you're on mac then: 

1. go to audio/midi setup
2. unplug your deluge
3. delete the deluge in midi setup
4. plug the deluge back in

closed 2024-11-16 by m-m-adams

todd-gochenour, 2024-11-16:
Attempting test now with 1.2 beta.   The new checkbox graphics in Deluge's menu are a much-appreciated improvement.  

Ableton Setup:
I'm using Windows.  Per previous comment, I do not see three ports for the Deluge in Ableton.  Ableton is configured to use the Focusrite USB MIDI audio controller for MIDI Input and Output ports.  Both Track and Sync options are checked for both of these ports.  Deluge MIDI output DIN routed to MIDI Focusrite Input. Windows PC connected via USB to Focusrite.  Focusrite MIDI output DIN routed back to Deluge Input.   It is a closed loop with two sequencers.

Tests:

| Play Master | Deluge Clock Input | Deluge Clock Output | Ableton Ext transport | Focusrite MIDI In Sync | Focusrite MIDI Out Sync | RESULT |
| ----- | ------------ | --------------- | -------------- | -------- | -------- | --------  |
| Deluge | ON | ON | ON | ON | ON | FAIL |
| Deluge | ON | ON | ON | ON | OFF | WORKS |
| Deluge | OFF | ON | ON | ON | ON | FAIL |
| Ableton | ON | OFF | OFF | ON | ON | WORKS |
| Ableton | ON | ON | OFF | ON | ON | WORKS |

The issue is with Ableton syncing MIDI output when it is a slave to the Deluge.  The Deluge clock settings are not a factor.  This is not a bug in the 1.2 beta.