SynthstromAudible/DelugeFirmware #2986: Recording Mode in Songview not working

Giansn, 2024-11-28:
### Please describe the problem:

I updated the firmware and now if I add a new audiotrack in songview, hold the record button so the belonging pad flashes purple, then press record and play it does show that its recording something, but after I press play again to stop the recording, all what was recorded just disappears and is not to be found or heard. 

### What is the expected behavior?

To be able to record into the purple flashing audioclip and hear the recording after it is finished.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

Chopin 1.2 

### If possible provide the steps to reproduce the issue and upload additional media:

1. Songview
2. add new audio track (audio1)
3. hold record button so pad belonging pad flashes purple 
4. press record and play 
5. start recording 
6. listen to recorded but recorded was not safed...

seangoodvibes, 2024-11-28:
@Giansn per the manual when recording to audio clips you don't press record to stop the recording, pressing record restarts the recording

To stop the recording and loop what you've recorded you need to press the clip launch pad

[image]

Giansn, 2024-11-29:
Thanks goodvibes! l think that should resolve it. If not l'll open a new issue you can close that one.

closed 2024-11-29 by seangoodvibes