SynthstromAudible/DelugeFirmware #2987: Any keyboard view + Record = Crash

panicmechanic, 2024-11-28:
### Please describe the problem:

As above on the latest nightly. 

### What is the expected behavior?

Recording.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

2024_11_28-165f8ab

### If possible provide the steps to reproduce the issue and upload additional media:

1. Open any song/create new
2. Enter synth clip
3. Open keyboard
4. Tap record
5. Tap play
6. Crash

panicmechanic, 2024-11-28:
If this is just me please lmk, maybe it's something to do with my SD card

m-m-adams, 2024-12-01:
I can't reproduce this on the latest nightly, I've tried various combos of quantization and count in bars as well as different types of synths

Can you confirm you are able to record on earlier nightlies without issues?

seangoodvibes, 2024-12-01:
> I can't reproduce this on the latest nightly, I've tried various combos of quantization and count in bars as well as different types of synths
> 
> Can you confirm you are able to record on earlier nightlies without issues?

It's reproducible for me.

Just went into keyboard view, turned on record and hit play.

seangoodvibes, 2024-12-01:
https://github.com/user-attachments/assets/de050cc2-9a18-4eda-97db-818b87b3a569

m-m-adams, 2024-12-01:
The bug is only present with "alternative playback start".  It seems to be miscalculating the position to resume from when you're in a clip with the keyboard open

closed 2024-12-01 by m-m-adams