SynthstromAudible/DelugeFirmware #2992: Since alpha 2024-11-24, a project from beta 1.2 is crashing when press play

nicolasbulb, 2024-11-30:
### Please describe the problem:

 I have a specific project called "Classic01" saved from the 1.2 beta latest version "2024-11-25" and Since alpha 1.3 version 2024-11-24, the project is loading but it crashes everytime you press PLAY

### What is the expected behavior?

The project should be converted and PLAY as expected.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2024_11_24-98586c7

### If possible provide the steps to reproduce the issue and upload additional media:

1. I have a specific project called "Classic01"  (included in the zip file) saved from the 1.2 beta latest version "2024-11-25"
2. Since alpha 1.3 version 2024-11-24, the project is loading
3. but it crashes everytime you press PLAY
4. The Deluge freezes with this state (included in the zip file)
5. I have to switch off the Deluge completely and if I get downgrade to either the latest beta "deluge-v1_2_0-beta+2024_11_25-e60ebac" or the alpha "deluge-v1_3_0-nightly+2024_11_23-8c8b146", it works.

[2024-11-30-Deluge-crash-alpha-2024-11-24-IMG_7623.zip](https://github.com/user-attachments/files/17966372/2024-11-30-Deluge-crash-alpha-2024-11-24-IMG_7623.zip)

m-m-adams, 2024-12-01:
In order to troubleshoot we'll need the collect all version of this song - press save, then hold select until you get a popup with the option to "collect media". That will put all the samples into a folder with the same name as the song

m-m-adams, 2024-12-01:
Oh actually I see from your crash code that this is a duplicate of #2987 which Sean has just fixed

m-m-adams, 2024-12-01:
If your bug isn't fixed in the next nightly build please reopen, this just fixes the crash that's in the image in your zip file

closed 2024-12-01 by m-m-adams