SynthstromAudible/DelugeFirmware #3014: Not able to learn a specific midi channel in a clip when connected by DIN.

Akevai, 2024-12-04:
### Please describe the problem:

In 1.1.1 when using the midi din (but not when using usb midi) when I learn an external midi device to a clip it does not learn the specific midi channel.  If I change channels on my midi device (or I am sending multiple channels  such as form a daw or another sequencer) the clip plays back all channels. This only happens with the din midi input. It behaves as expected with the usb midi input. 

### What is the expected behavior?

When leaning a midi inout it only learn the channel used when leaning so if data is one on a different channel that clip doe not respond. This means for example you can send channel 1 to bass and channel two to a pad and switch  channels on your midi keyboard to play the different instrument.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

C1.1.1

### If possible provide the steps to reproduce the issue and upload additional media:

1: Deluge on
2: Connect midi sequencer to din input
3: Open new synth clip (S1)
4: Mute all channel except 1 on the external  device
5: Learn that channel to the S1 clip
6: S1 plays
7: Mute channel on ext  sequencer 
8: Unmute channel 2 on ext sequencer
9: S1 plays (it should not and does not when using usb midi instead of din)
10: open a new synth clip (S2)
11: Learn external sequence channel 2 to that clip (with all other channels muted) 
12: S1 and S2 both play ch 2
13: Unmute channel 1
14: S1 and S2 play both channel 1 and channel 2

seangoodvibes, 2024-12-04:
@Akevai can you confirm if this happens on our 1.2 beta please?

Akevai, 2024-12-04:
Yes it still does unfortunately

seangoodvibes, 2024-12-04:
> Yes it still does unfortunately

Do you have midi follow enabled?

Akevai, 2024-12-04:
No I turned it off when I first notice the problem to check if that was why

m-m-adams, 2024-12-11:
It's probably MPE settings, check in the midi device's menu to make sure that upper and lower zones are zero for DIN

closed 2024-12-14 by m-m-adams

Akevai, 2024-12-31:
Excellent that indeed solved the problem. Thank you also much for you amazing work and quick support.