SynthstromAudible/DelugeFirmware #3038: Mild weirdness if you accidentally press select while turning it to set row probabilityp

abstractrealism, 2024-12-10:
### Please describe the problem:

Was setting the iterance for a row with audition + turning select right when I inadvertently pressed select while turning and the audio glitched out. The root notes disappear from the audition column though scale mode is still engaged, and the audition pads jumped down to the -2 octave. Pressing “Scale” twice fixes it. 

### What is the expected behavior?

It was an accident so not sure what would be expected, probably nothing. If scale mode is off nothing happens besides blinking the scale button. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2.0 Beta DD19D1F

### If possible provide the steps to reproduce the issue and upload additional media:

1. With scale mode on, in a synth clip, press and hold an audition pad. 
2. Then press the select encoder and turn right.

seangoodvibes, 2024-12-10:
@abstractrealism this is not a bug but a feature.

It changes that note row's note (you can see the change on the display) eg D#-1 to D-1 to C#-1

abstractrealism, 2024-12-10:
@seangoodvibes In my case it is changing C3 to B-2 though, and when turning right which I think should go to C#3?

seangoodvibes, 2024-12-10:
> @seangoodvibes In my case it is changing C3 to B-2 though, and when turning right which I think should go to C#3?

Pretty sure it depends on your scale

I was just using the regular default scale

abstractrealism, 2024-12-10:
> > @seangoodvibes In my case it is changing C3 to B-2 though, and when turning right which I think should go to C#3?
> 
> Pretty sure it depends on your scale
> 
> I was just using the regular default scale
> 
> Probably not every note changes by 1 semitone?

It’s happening for me in a brand new song default template, scale, etc, see attached. 

Will edit the title to reflect that this is a feature bugging out rather than 100% bug. (I don’t really use custom scales so I forgot about changing a note row that way). 

https://github.com/user-attachments/assets/36e2cad0-c37a-435e-b422-5902dc83bea7

seangoodvibes, 2024-12-10:
> It’s happening for me in a brand new song default template, scale, etc, see attached.
>
>
>
> Will edit the title to reflect that this is a feature bugging out rather than 100% bug. (I don’t really use [...]

Ah you're right there is a bug, or at least it feels like a bug

For some reason it jumps from a high note to a low note

Can you check how it works on official or earlier releases and report back?

Thanks

seangoodvibes, 2024-12-10:
@nikodemus  could this be possibly related to the scale changes?

nikodemus, 2024-12-10:
> @nikodemus could this be possibly related to the scale changes?

I'm not going to say no, but can't think how it would. I can take a look.

seangoodvibes, 2024-12-11:
> I think there's a bad interaction with https://github.com/SynthstromAudible/DelugeFirmware/pull/2364
> 
> 
> 
> Need to think a bit, but I don't think is a deep bug.

Ah I wonder if it's also the cause of this issue:

https://github.com/SynthstromAudible/DelugeFirmware/issues/2936

nikodemus, 2024-12-11:
> > I think there's a bad interaction with #2364
> > Need to think a bit, but I don't think is a deep bug.
> 
> Ah I wonder if it's also the cause of this issue:
> 
> #2936

Super plausible! I'll put that on my queue as well.

PR #3054 (2024-12-11): fix sharpening/flattening scale notes
- ScaleChange was missing the target scale, used to figure out a possible y-adjustment. Fixes #3038.

- Same issue in rotateMusicalScale(), used from Song::transposeAllScaleModeClips, which might be causing #2936.
commit 39dfd70e (2024-12-11): fix sharpening/flattening scale notes (#3054)

- ScaleChange was missing the target scale, used to figure
  out a possible y-adjustment. Fixes #3038.

- Same issue in rotateMusicalScale(), used from
  Song::transposeAllScaleModeClips, which might be causing
  #2936.
PR #3055 (2024-12-11): fix sharpening/flattening scale notes
- ScaleChange was missing the target scale, used to figure out a possible y-adjustment. Fixes #3038.

- Same issue in rotateMusicalScale(), used from Song::transposeAllScaleModeClips, which might be causing #2936.
commit 579da08c (2024-12-11): fix sharpening/flattening scale notes (#3054)

- ScaleChange was missing the target scale, used to figure
  out a possible y-adjustment. Fixes #3038.

- Same issue in rotateMusicalScale(), used from
  Song::transposeAllScaleModeClips, which might be causing
  #2936.
closed 2024-12-11 by m-m-adams