SynthstromAudible/DelugeFirmware #3058: Ocs Sync issue

Qui-Gon-JooL, 2024-12-11:
### Please describe the problem:

 Hello,
for several nightly builds the OSC Sync no longer works, instead a noisy sound.

### What is the expected behavior?

It should work

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

24-12-11 nightly build

Qui-Gon-JooL, 2024-12-14:
SYNC mod still not working on deluge-v1_3_0-nightly+2024_12_14-20ad4c9.bin

seangoodvibes, 2024-12-15:
@Qui-Gon-JooL please use the beta build for now. This will be fixed whenever it gets fixed, sorry but it's the nature of the nightly build, as much as try prevent things like this from happening. The nightly is "experimental" at the moment.

PR #3102 (2024-12-20): Fix sync issue (variable writeback)
Fixes #3058
closed 2024-12-20 by seangoodvibes