SynthstromAudible/DelugeFirmware #3066: Turning vert. Encoder in velocity view cuts off notes

pssssssst, 2024-12-15:
### Please describe the problem:

When you turn the vertical encoder in velocity view it sometimes cuts off notes

### What is the expected behavior?

Turning vertical encoder doesn't interfere with the notes playing

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

beta 14dez24

### If possible provide the steps to reproduce the issue and upload additional media:

1. Open synth clip.
2. program some longer notes
3. press play
4. enter velocity view
5. turn vertical encoder fast

seangoodvibes, 2024-12-15:
Thanks for reporting

I don't think we'll have a solution for this for the beta other than:

- not turning the vertical encoder "fast" / turn it slower
- not turning the vertical encoder while playback is running

I'm working on some UI improvements that may help but it won't be part of the 1.2 release.

pssssssst, 2024-12-15:
No problem.
It doesn't bother me in my workflow at all.

m-m-adams, 2024-12-26:
Fixed by #3126

closed 2024-12-26 by m-m-adams