SynthstromAudible/DelugeFirmware #3083: Scrolling horizontally rapidly in Waveform view while auditioning a kit row will cull sample playback early.

ok-reza, 2024-12-17:
### Please describe the problem:

Load a drum loop into a kit row and enter waveform view.
Zoom in a little bit into the waveform and while auditioning the pad (holding the audition pad down) start scrolling horizontally back and forth, the playback of the sample will cut off early. If you aren't able to reproduce, adjust zoom level, sometimes too far zoomed in or too far zoomed out will not cause the cull.

Tested compared to 1.1.1 and no culling occurs there.

This has larger implications specifically when slicing samples, where you normally would be constantly auditioning pads while also scrolling through the waveform to locations where you want to adjust the start/end and loop points.

### What is the expected behavior?

No culling while auditioning a sample and moving horizontally through the waveform

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2-RC1

ok-reza, 2024-12-17:
https://github.com/user-attachments/assets/8e6a207e-c80a-425e-bcd8-5f9a3fd94afa

ok-reza, 2024-12-17:
recreate easily with 1 kit clip and 1 kit row:

1) new project
2) create new kit
3) navigate to ARTIST>PHIL ELVERUM
4) load 'GUITAR OCEANROAR'
5) set sample 1 mode to ONCE
6) enter waveform view
7) zoom in at least twice
8) tap audition and scroll via <> and you'll hear the sample end abruptly instead of playing all the way through

seangoodvibes, 2024-12-17:
Bisected and confirmed that the bug originates from https://github.com/SynthstromAudible/DelugeFirmware/commit/dac64cec38434155c94e013331c9a82e50cf952d

closed 2024-12-17 by stellar-aria