SynthstromAudible/DelugeFirmware #3107: Voice and performance problems

klangweber, 2024-12-21:
### Please describe the problem:

Hello Community 🎹 The performance topic still leaves me somewhat puzzled. After updating to 1.2RC, the CPU limit seems to be reached more quickly. Here’s an example: I’ve already resampled almost all tracks containing multisamples into audio clips. By the way, I’ve also consolidated and resampled drum tracks.
Now I want to resample this multisample pad heard here, which consists of two-tone chords. It’s a long sound with a lot of release… all other tracks were muted, and there’s nothing unnecessary left in the song. Everything was deleted except for the resampled audio tracks, a few kits, and some MIDI tracks. After about 10 attempts, I finally managed to resample this pad without voice stealing.
Before the update, the sounds were already pushing the limits, but I could still relatively comfortably fire off the track as a multisample in the arrangement. Now, the CPU is hitting its limit during resampling.
In the end, it worked, and I was able to achieve my goal with this small project using the audio clips. The arrangement was played back with multiple audio clips, external synthesizers, and drum kits and I have recognized only one voice struggle whilst recording.

### What is the expected behavior?

I have expected that the CPU can handle it without any struggle after resampling nearly all tracks to audioclips (4 audio, 2 kit, one midi for external) with just subtle usage of delay, reverb, sidechain - the older version had 3 multis and several kits ...

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2rc

### If possible provide the steps to reproduce the issue and upload additional media:

Song with audioclips and less struggle: https://drive.google.com/drive/folders/12y7alTZcPEyxYl__no4QDgaDXAHY-mZH?usp=drive_link

Older version with multis and hard struggles: https://drive.google.com/drive/folders/1x37aCTISYv5BibRJBDKbxV2ocoaRwuEG?usp=drive_link

m-m-adams, 2024-12-22:
I requested access to your files since they're private. You can also upload a zip directly to the post if you prefer

klangweber, 2024-12-22:
Hey, does it work now?
Greets
Ben

m-m-adams, 2024-12-22:
The "few kits" are filling your memory - there's a ton of sounds loaded that aren't actually in use. You'll get much better performance if you consolidate the sounds you use into one kit 

It's performing slightly worse than 1.1 because the binary is a bit larger and that's not something we can fix in this release, though we are constantly working on lowering the size

klangweber, 2024-12-22:
Thx for your Support. What about the second song version with resampled
audioclips? In had problems to resample the padsound, voiced dropped and I
needed several trys to catch a record without cpu blinking. Only one Multi…
everything else was muted (other 5 tracks)

Greets from Hamburg - Germany

Ben

m-m-adams, 2024-12-22:
Sorry if you're doing this over email you might have missed the end of my comment, when I try it plays all the way through without problems so you might want to pickup a better SD card. There's a list of recommended ones in the wiki but the best is the Kingston Canvas Go Plus

klangweber, 2024-12-22:
Oh sorry… yes I replied just per mail. Ok I have ordered the card. Thx for your help. I hope the Performance can be improved in the future. For now I will use fewer kits and also external synth for long pads. 

I wish you a wonderful Christmas Season and relaxing days.

closed 2024-12-26 by m-m-adams