SynthstromAudible/DelugeFirmware #3133: Recorded audio clips not visible in file browser before saving song

pssssssst, 2024-12-24:
### Please describe the problem:

When i'm recording an audio clip i can only load that clip in a kit or another audio clip after i have saved the song. It doesn't show up in the file browser before. 
When i resample i can immediately go to the resample folder to load that audio file.
Which is probably how it should work.

### What is the expected behavior?

I think it should be visible in the file browser immediately.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

 deluge-v1_2_0-rc1

### If possible provide the steps to reproduce the issue and upload additional media:

- record audio clip
- make new init kit (for example)
- try to load just recorded audio clip in a kit row

ok-reza, 2024-12-25:
this is expected behavior. you have to save to get recordings saved permanently. i believe they're in a hidden temp folder prior to that (and would get deleted on power down)

pssssssst, 2024-12-25:
I was just wondering, because resampled audio files behave differently and are available in the file browser right away.

m-m-adams, 2024-12-26:
Intended behaviour, it avoids live looping cluttering the sd card

closed 2024-12-26 by m-m-adams