SynthstromAudible/DelugeFirmware #3148: Audio Clips Shift

Evilpilli, 2024-12-25:
### Please describe the problem:

Sometimes, but not concistenlty when loading up and old song. Certain audio clips will have changed, and an old audio clip is now occupying the space where the original is supposed to be. The original is gone, and the wave file has been duplicated, but onto the original audio-clips name. 

### What is the expected behavior?

For audio clips (samples) not to shift around when loading songs

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### If possible provide the steps to reproduce the issue and upload additional media:

Not entirely sure how to reproduce it unfortunatly. I'm doing a clean of my SD-card come new year, and I'll report back if that solves the issue.

nikodemus, 2024-12-30:
Possibly related to https://github.com/SynthstromAudible/DelugeFirmware/issues/3018

seangoodvibes, 2025-05-28:
@Evilpilli did you ever encounter this issue again?

Thanks