SynthstromAudible/DelugeFirmware #3158: Zooming in and out while playing song creates pops

Tsapio, 2024-12-26:
### Please describe the problem:

I double checked and this was not an issue in version c1_1_0.  When I play the attached song in 8th-notes and either zoom in to 16th or out to 8th, there are audible pops.  This doesn't exist in all my songs but many of them.  

### What is the expected behavior?

The expected behavior is to zoom in and out without any audible pops.

### When did you start noticing the issue?

As soon as I updated to c1_2_0 from c1_1_0.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turned on, 2. Navigated to the song "Build a Bridge", 3. Pressed play, 4. Zoomed in and out from 8th-note view to 4th and 16th, etc.
[Build A Bridge.XML.zip](https://github.com/user-attachments/files/18250461/Build.A.Bridge.XML.zip)

m-m-adams, 2024-12-26:
Can you make a collect all save of the song and share it?

m-m-adams, 2024-12-26:
Also would you mind checking if it's still an issue on the nightly?

Tsapio, 2024-12-26:
I verified it is also in the nightly build, yes. Sorry, I should have done that first.  Here you go!

[Build A Bridge 2.zip](https://github.com/user-attachments/files/18255101/Build.A.Bridge.2.zip)

SaschaRadojkovic, 2024-12-27:
i had the same problem mostly with audioclip projects .. and i now tried the release candidate  i used before because i was shure not to have this problems [https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/release_1_2_0_rc1](url)  and yes there i dont have the problem.

sapphire-arches, 2025-02-08:
If this is happening in row view, this is likely a duplicate of #3177

Tsapio, 2025-02-09:
@sapphire-arches yes, it looks like it is a duplicate and is still happening. Thank you

seangoodvibes, 2025-05-15:
@Tsapio can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

Tsapio, 2025-05-16:
@seangoodvibes I just downloaded 1.2.1 and can confirm in my testing the pops are gone!  I will keep testing to make sure but at first glance, things sound great! Thank you!

seangoodvibes, 2025-05-16:
Per comment above, closing this issue as it appears it was resolved!

closed 2025-05-16 by seangoodvibes