SynthstromAudible/DelugeFirmware #3161: Audio glitches and crashes on songs with Audio tracks.

sissyfus, 2024-12-26:
### Please describe the problem:

I get white noise with echo on songs with audio tracks, especially when loading while playing. The Deluge also crashes when playing these tracks. 

### What is the expected behavior?

The tracks play normally on community firmware 1.0

### When did you start noticing the issue?

When I tested Chopin 1.2 firmware

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

deluge-c1_2_0.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Load a song with an audio track while playing. 
[image]

https://github.com/user-attachments/assets/ebb57eb3-6ebb-4486-b0ed-e50140370310

m-m-adams, 2024-12-26:
Please share a song that shows these bugs. I am not able to reproduce and many people use songs with audio clips without issues

sissyfus, 2024-12-26:
I load this song and start scrolling the other songs and it happens: 
[SONG033A.zip](https://github.com/user-attachments/files/18254745/SONG033A.zip)

sissyfus, 2024-12-26:
> Please share a song that shows these bugs. I am not able to reproduce and many people use songs with audio clips without issues

This one crashes the Deluge straight without letting me play or save: 
[SONG038.zip](https://github.com/user-attachments/files/18254767/SONG038.zip)
[image]

m-m-adams, 2024-12-26:
You need to make a collect all save of the song and upload the whole thing, the file you shared doesn't have the audio files included

sissyfus, 2024-12-26:
This one is just a loop. It crashes the Deluge most of the time: 
[33A_2.zip](https://github.com/user-attachments/files/18254847/33A_2.zip)

m-m-adams, 2024-12-26:
That one is missing the song xml file

sissyfus, 2024-12-26:
Here!
[33A_2.zip](https://github.com/user-attachments/files/18254907/33A_2.zip) 
Now I'm starting to believe it's a SD-card issue and will try to with another card!

m-m-adams, 2024-12-26:
Ok great thanks! I'll give this a test later today and see if I can spot the problem

sissyfus, 2024-12-26:
Thanks a lot, thought I upgraded to a better SD-card than the original one from 2017 but the Deluge seems to be pretty picky on cards! Maybe the error codes can tell if it's just that and not the firmware.

pulzarTZ, 2024-12-27:
Hey Guys!
I've started typing up a report but I'll just hop in here as my issues are kind of the same. Below copied.

My Deluge freezes on random song changes. On Beethoven it mainly happened when I changed songs holding down Tempo for beat matching. On Chopin it happens even when it's not playing and I just change a song after startup. It happens randomly, but definetly more frequently.
There are certain songs that are more prone to produce these issues but couldn't figure out a conncetion. On the pictures of the crash/freeze state there is one with more colored pads, which was on Beethoven, and the red blue one happens on Chopin. The latter has the same freeze state as at least one other user.
As I'm typing this up, the blue/red crash happens on almost every song change now. I'm using the SD card for three years now and the issues started with Beethoven adn Chopin especially, but the song files and the folder strucutre is the same ever since. Might that be an issue? Because I've made a lot of songs pre community firmware. If that's the issue what

[image]
[image]

https://github.com/user-attachments/assets/ea343c61-8ba2-45de-8b3c-e5508d461b97

 would be the best procedure to format the card and load everything back, so it's the most compatible with Chopin?

m-m-adams, 2024-12-28:
@pulzarTZ please make a new issue since this looks unrelated

sapphire-arches, 2025-02-08:
Filed #3350 for @pulzarTZ's issue.

@sissyfus do you still see crashes with the new SD card?

sissyfus, 2025-02-09:
@sapphire-arches Changing to a Kingston Canvas Go solved the issue, I still occationally get crashes when creating new songs, 
but I'm not sure if that might be something else. Seems like the 1.3 can't handle slow SD cards as well as the official firmware? 

> Filed [#3350](https://github.com/SynthstromAudible/DelugeFirmware/issues/3350) for [@pulzarTZ](https://github.com/pulzarTZ)'s issue.
> 
> [@sissyfus](https://github.com/sissyfus) do you still see crashes with the new SD card?

seangoodvibes, 2025-05-15:
@sissyfus can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

sissyfus, 2025-05-16:
For me changing to a Canvas Go SD-card solved this - I'll have to wait to
stable 1.3 to check it since I use the Deluge on stage
but will check it when I come so far!

Great work mates!

-- 
Jørgen Skjulstad

<http://www.soundcloud.com/counivers>https://linktr.ee/counivers
https://www.centeroftheuniver.se/
Subscribe on Youtube
<https://www.youtube.com/sissyfus?view_as=subscriber?sub_confirmation=1>:
Mastodon ***@***.***>
Bluesky <https://bsky.app/profile/counivers.bsky.social>
TikTok ***@***.***>
Facebook <https://www.facebook.com/counivers>
Spotify <https://open.spotify.com/artist/2wO5N1U2VYDZzs4gTSNVMm>
Soundcloud <https://soundcloud.com/counivers>
Instagram <https://www.instagram.com/counivers>
Bandcamp <https://meau.bandcamp.com/>

Label: Metronomicon Audio <http://www.metronomiconaudio.net/>
Join the Metronomicon Audio mailing list
<https://www.metronomiconaudio.net/>:

seangoodvibes, 2025-05-16:
Per comment above, closing issue as I believe it was resolved. We can revisit again later if the issue persists.

closed 2025-05-16 by seangoodvibes