SynthstromAudible/DelugeFirmware #3164: Audio Monitoring Channel in Arranger View Does not retain Gold Knob Values after save/reload

OaklandGhosts, 2024-12-26:
### Please describe the problem:

When I create an Audio Channel and set it to Monitoring in Arranger view (I typically use Stereo Input/Monitoring) and change gold knob values such as Volume, Pan, Cutoff, delay, reverb, etc - if I save the DOng and return to it later, those values are not retained and return to default. This does not happen with Audio tracks I record to or load samples to, only monitoring tracks set up in Arranger View (that don't appear in Song/Grid View)

### What is the expected behavior?

I expect the values I set to be retained and reloaded after a save.

### When did you start noticing the issue?

I have seen this issue come and go across various community builds. I first saw it in 1.1, mentioned it and saw it repaired in a later build. Saw it appear again with build 1.2 beta (1.2.0-Beta-175964E) sounded like it was possibly fixed in a later beta build. Thought it worth mentioning as It appears to be an issue that comes and goes across various builds. Can confirm the issue is present on the full release Build of C1.2.0 launched yesterday. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

C1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

step 1, go to arranger view and create an audio track. step 2, press learn and the audio track to set the input. step 3, set input to Stereo (monitoring) Step 4, change various gold knob values to taste. Set 5, save song. Step 6, lad a different song. Step 7 reload previous song - Values set on gold knobs for Audio Monitoring Track in Arranger view are back to default values and not user set/saved values.

ok-reza, 2024-12-26:
i wasn't able to reproduce this UNTIL i made a new audio track while in arranger view (so don't convert a clip already made and accounted for in SONG view). it's important that the clip made doesn't show up in SONG view, which is only possible when making a clip while in arranger view.

while this is investigated/fixed, the workaround in the meantime would be to create the audio clip w monitoring on while in SONG view then switching over to arranger. this will ensure the values you set don't reset when reloading the project.

OaklandGhosts, 2024-12-26:
The problem with creating an Audio track in song view and setting it to monitor, is that if I record anything on any other track - say, a live keyboard sequence on an internal Synth track, or some FX automation on a drum track, etc - it will record the incoming audio to the Audio Track. This isn't the behavior I'm after. I just want to monitor the input, not record it. The idea is to use the input on the deluge as an extra I/O in situations where my mixer is full. 

Is there another way of achieving this I might not be aware of? I was told a very long time ago about the creating an audio track in arranger view trick to achieve this, and it works as intended- aside from the settings no longer being saved.

An alternative solution that might be worth investigating in a future build is making it so that if an audio track is set to monitoring mode it just monitors and doesn't record to the track? Or perhaps only letting the audio track record audio if I am in the audio track (not when I'm in another synth clip recording a sequence for example)

ok-reza, 2024-12-26:
you need to disarm the audio clip you make in song view so then it's not affected by any future recordings. in song view i believe you hold record down and press the mute/unmute pad of the audio clip until it's green (disarmed).

i think a new approach to audio clips has been created for the next firmware 1.3 which may do what you suggested at the end of your post actually

OaklandGhosts, 2024-12-26:
Nice! hopefully that makes it a little more straightforward. 

In the meantime your above workaround for disarming the audio clip works so thank you. Feels a bit hidden feature-y but if it works it works! Thanks man!

sapphire-arches, 2025-02-08:
It sounds like this is intentional behavior/a documentation gap? Please reopen if that is not the case.

closed 2025-02-08 by sapphire-arches