SynthstromAudible/DelugeFirmware #3165: Long text occasionally doesn't scroll on OLED 1.2 Chopin

MkUltra85, 2024-12-26:
### Please describe the problem:

When the track title is too large to fit on the OLED screen, it will occasionally not scroll to show all the text. I noticed this both on the new track title added to the main screen in Song / Arranger mode, and in the track loading screen.

It has been very inconsistent, and comes and goes even without power cycling or reloading the project; sometimes it is frozen with the text cut off, and sometimes it scrolls as expected. 

I'll continue to play with it and update here if I can find any trigger that causes it regularly.

### What is the expected behavior?

Scrolling when the text cannot fit on the OLED.

### When did you start noticing the issue?

Since updating to the 1.2 Chopin firmware today.

### Is there a relevant Pull request?

I didn't see one

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.0

ok-reza, 2024-12-27:
scrolling has been disabled during playback. this was partly due to scrolling getting laggy and i believe making the deluge use more of its resources on the display instead of processing sounds and also because some said the scrolling was distracting.

open to feedback as i don't think the perfect solution may have been met just yet

MkUltra85, 2024-12-27:
Thanks, Reza. I tried to look, but didn't see anything noted about this. I tend to err on the side of making silly posts rather than not posting problems. 

It would be useful to at least be able to scroll through the text in the song loading screen with <>.

closed 2024-12-28 by m-m-adams