SynthstromAudible/DelugeFirmware #3169: Crackle on save while playing

jannezz, 2024-12-28:
### Please describe the problem:

If you save a song while it's running, there's an audible crackle IF there's a KIT track present.

### What is the expected behavior?

No crackle!

### When did you start noticing the issue?

First on 1.2-RC1 now on official 1.2. Doesn't happen on 1.1. with the same song file. Tried two different SD cards. Crackle is heard every time when SAVING.

### Is there a relevant Pull request?

Dunno.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

C1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on.
2. Add notes to default track 
3. Play
4. Create a KIT track
5. Save (CRACKLE!)

nourileiner, 2024-12-28:
I noticed the same behavior. 

It wasn't present on 1.1, only since I installed 1.2

m-m-adams, 2024-12-28:
We're aware, it's a result of allowing songs to get closer to the CPU limit in 1.2 than 1.1

jannezz, 2024-12-29:
Good that you are aware!

Bummer that I have a habit of saving even when playing live and I really dig the 1.2. firmware, but this one tiny thing makes it unusable for me. 

Well, can't have it all.

You're doing amazing work anyways!

m-m-adams, 2025-01-07:
Could you check 1.3?

jannezz, 2025-01-07:
No crackling on 1.3.0-Nightly-2AD07A5!

At least on the couple of songs I checked that had issues on 1.2.0.

Thank you for the fix!

closed 2025-01-07 by m-m-adams