SynthstromAudible/DelugeFirmware #3174: Sd card contents vanished when loading Chopin 1.2 firmware

jomdonsen, 2024-12-29:
### Please describe the problem:

Hi, also just posted this on the facebook group page - I
[image]
 just attempted the Chopin 1.2 firmware update (following along with the directions via GitHub) and after backing up my sd card (in three different places, fortunately) and deleting my old firmware .bin file, v4.1.4), I attempted to put the Chopin 1.2 .bin on my card but got an error message that it could not be read and then the entire contents of my sd card vanished. 

When I go to put the original copied files back on my sd card, I get this message below. Very freaked out! Appreciate any guidance, thanks. Also, the card won’t eject with the message that “one of more programs may be using it”, but I am reluctant to force eject.

### What is the expected behavior?

That the new firmware would install

### When did you start noticing the issue?

Just now

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

nikodemus, 2024-12-30:
This sounds more like a corrupted SD card than anything else.

m-m-adams, 2025-01-07:
It looks like your card is corrupt. I'd just reformat it and then copy your old files over. For best performance you'll need to use FAT32 with 32kbyte clusters when you format it

closed 2025-01-12 by m-m-adams