SynthstromAudible/DelugeFirmware #3177: Audio glitches/drop outs when scrolling song view

rudolphrapid, 2024-12-29:
### Please describe the problem:

After installing 1.2.0 release I get sound drop outs and clicks/glitches in my projects when playback is active and I scroll song row view up and down or left and right. I continuously get this issue with old and new projects.

### What is the expected behavior?

I expect not to get any sound issues while scrolling the song row view.

### When did you start noticing the issue?

Right after installing 1.2.0 Chopin update.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

Please download a sample project from here: https://drive.google.com/file/d/18svnoOjTsmGwrcf7lnvoXA45GPcTfFIz/view?usp=sharing

It's an old project far older than the community patch. Load it with 1.2.0 Chopin version and switch to song row view. Find either the yellow or the red section and unmute it and start playing the song. While it's playing scroll the song view up, down, left, right. You can here some drop outs and many sound glitches like short strange trumpet sounds.

This issue is not reproduceable with 4.0.1 official firmware.

m-m-adams, 2025-01-07:
Can you try the nightly? We've made several performance improvements 

If you do just be warned not to save your song unless it's backed up, songs made in the current nightly can't be loaded by 1.2 or earlier firmwares

rudolphrapid, 2025-01-07:
I've tried it with the latest nightly build and still have the very same experience.

There's a comment on this ticket stating that song had been allowed to get closer to CPU limit than before: https://github.com/SynthstromAudible/DelugeFirmware/issues/3169

Can this issue also be related? If so is it possible to somehow revert that change and release a fix for 1.2? The latest version of community firmware has a lot of very cool features but this performance glitch makes it unreliable currently.

m-m-adams, 2025-01-07:
I don't think so, there must be something specifically weird about rows mode song scrolling

beachbento, 2025-01-13:
Actually the audio glitch seems to be applied to any audio recording. If I have a current audio loop going, with say a guitar loop and one drum kit, any additional liveloop recording has the clicky cpu like pop sound.

seangoodvibes, 2025-05-15:
@rudolphrapid can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?