SynthstromAudible/DelugeFirmware #3191: Degradation in looper audio quality

safrosch, 2025-01-02:
### Please describe the problem:

I'm having two separate issues with the looper functionality. 

(1) Whenever I am in song view (rows) and I am creating an audio loop with no other tracks running, the first 2-3 seconds of the loop will sound degradated, very similar to if I were to turn up the decimation effect. I've recreated this with a number of different audio signals feeding into the audio input as well as using the onboard microphone. When the loop plays back the degradation is gone, it only appears in the monitored sound during the looping process. Subsequent overdubs (loop and layer) do not have this same issue with the monitored sound. This also occurs for all methods of monitoring incoming audio (left, right, stereo, and balanced). Note, this issue does not occur when creating an initial audio loop in grid mode. 

(2) When looping in grid mode (which, by the way, I love this feature) every time I try to create overdubs both on loop & layer mode, the total output of the track drops significantly. I noticed that unlike when overdubbing in row mode, overdubbing in grid mode merges overdubs and the initial loop into one audio track, like more traditional loopers. I'm not sure if the output level drop is intentional so the new audio track doesn't clip with superimposed loops, but at some points it's dropping to a near inaudible level. 

### What is the expected behavior?

Ideally the incoming monitored sound should be crystal clear when looping. Again this is only an issue on the primary loop, all overdubs don't have this issue. 

### When did you start noticing the issue?

Immediately after downloading the new Chopin 1.2 firmware. The issue of monitoring sound degradation during looping was not an issue in the Amadeus or Beethoven firmware updates. 

### Is there a relevant Pull request?

N/A

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

Chopin 1.2

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on Deluge and open song mode (rows). 
2. Create an audio track with active monitoring (does not matter if it's left, right, stereo, or balanced). 
3. Ensure audio track & deluge are armed to record and hit play. 
4. While this track is recording ensure audio is being fed into the deluge via the 1/4" stereo input or via the onboard microphone. Note, if no sound is being fed, the deluge will not feedback or create noise. The issue only affects incoming audio and only for the first 2-4 seconds of recording. Subsequent playback is unaffected. 

To create the second issue I posted about: 
1. Turn on deluge and open song mode (grid view). 
2. Create an audio track with active monitoring. 
3. Arm audio track and deluge to record and press play. 
4. Play or sing into the deluge to ensure audio is being captured in this initial loop. 
5. Using either an external midi controller or the new red & pink buttons in grid view, start to overdub, once again playing or singing into the deluge. 
6. The combined loop and overdub will now be playing back at a much lower volume than they were recorded at.

m-m-adams, 2025-01-07:
Hey yeah because of how the volume control works each loop will get quieter unless you have it at max (it lowers the volume pre recording).

A few other people have mentioned this as an annoyance so it would be nice to adjust how the volume knob works. I just set the volume using the control on whatever I'm recording for the moment

m-m-adams, 2025-01-07:
Oh and I have no idea on that row mode looping bug

baymud, 2025-01-09:
I just ran into the glitchy audio issue too. As described it occurs reliably when there are no other tracks (ie, starting a new project by recording into an audio clip). The first few seconds will sound glitchy/full of artifacts, almost like a buffer is trying to catch up or something, but the glitches aren't recorded

baymud, 2025-01-09:
For what it's worth, I can also reproduce the audio glitch in 1.3

beachbento, 2025-01-13:
I’m having the same issue. When live looping, the audio quality just has a weird buffer glitch whenever I am recording a new loop. This is in song mode rows mode.

m-m-adams, 2025-01-13:
Is anyone having the start of loop issues with a Kingston canvas go plus sd card? I'm only able to replicate it with rows mode and the stock sd, either grid mode or a good sd card seems fine

baymud, 2025-01-13:
> Is anyone having the start of loop issues with a Kingston canvas go plus sd card? I'm only able to replicate it with rows mode and the stock sd, either grid mode or a good sd card seems fine

Yes, I can replicate the glitch with both my new Kingston Canvas Go Plus and Sandisk Extreme cards on both 1.2 and the latest Nightly, but no issues on 1.1.1

m-m-adams, 2025-01-17:
Oh ok this is not at all what I thought the issue was! There seems to be a separate bug where the recording of the audio file can end up slightly glitched even though the monitoring sounded normal, this is the opposite

I'll have another go later and see if I can get this to happen since I haven't noticed it before

seangoodvibes, 2025-05-15:
@safrosch can you confirm if you still have this issue with our 1.2.1 release / latest 1.3 beta release?

baymud, 2025-05-15:
@seangoodvibes the first part of this ticket is the audio loop glitches I've been talking to you about and they are fixed in 1.2.1/1.3. The second part seems to be intentional in the way overdubbing works, as per @m-m-adams, so I think this issue is all sorted

seangoodvibes, 2025-05-15:
Closing per @baymud’s comment above. We can reopen if there is still an issue that needs addressing, but I think it’s solved.

closed 2025-05-15 by seangoodvibes