SynthstromAudible/DelugeFirmware #3210: Copying MIDI clip = crashes on save

panicmechanic, 2025-01-04:
### Please describe the problem:

Copying the midi clip in the attached songs will cause the deluge to hang or crash.

### What is the expected behavior?

To save.

### When did you start noticing the issue?

Today.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

nightly-598cd8e

### If possible provide the steps to reproduce the issue and upload additional media:

[Bug.zip](https://github.com/user-attachments/files/18307066/Bug.zip)

Please open either song, copy the midi clip, then try and save. 

Each song has a different response to the bug, the only different is the name of each file. 

For the record, I didn't add the "3:" to the "no crash screen" song, it happened after a previous crash. Might be the cause?

m-m-adams, 2025-01-07:
Give this another try, we fixed some bugs around naming things!

panicmechanic, 2025-01-07:
Bug persists on the save file I shared, but has changed. Now it crashes immediately upon copying.

I also have a new track on the most recent nightly doing the same thing.

panicmechanic, 2025-01-07:
This doesn't happen on brand new tracks. I.e. if I make a new track, add a midi clip, add some notes, and copy it - no problem. Saves fine.

panicmechanic, 2025-01-07:
Lemme know if having another song save file with the problem would be helpful, and I'll upload it

seangoodvibes, 2026-06-14:
I tested your songs with current nightly and cannot reproduce. Please reopen issue if it is still a problem.

closed 2026-06-14 by seangoodvibes