SynthstromAudible/DelugeFirmware #3224: CFW 1.2.0 - Number of clips in Clip View and Row View do not match

viessg, 2025-01-06:
### Please describe the problem:

When working in Grid View for a while and switching to the Row View, I noticed huge number of various strange tracks lots and lots more than I have in Grid View. In Grid View I had 9 clips, but in Row View there are dozens of rows and clips absolutely not groped by colours - in fact all mute pads are red and all audition pads are cyan.
Could it be that - when I delete clip in Grid View by Clip+Shift+Delete (in Blue mode) the clip doesn't gets removed in Row View and stays there as loads of trash? Shouldn't the deletion of clip/row be synced in both modes?

### What is the expected behavior?

The number of clips in Grid View and in Row View should match. If I remove/delete clip from Grid View it should be removed/deleted also in Row View and vice versa

### When did you start noticing the issue?

First time when I tried Grid View

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

Song file is attached. Grid view is clean, but Row View is populated with trash clips.
[Cfw Test 2.zip](https://github.com/user-attachments/files/18322050/Cfw.Test.2.zip)

m-m-adams, 2025-01-07:
Were you looping clips? If so this is what happens. The clips exist in both modes but only one clip per track can play at a time grid mode shows the active one (should be the cyan and green clip in rows mode). We're not going to change this, it's possible to represent more clips in rows mode than grid mode. We've aimed to minimize the impact but the extra clips are essential for some songs and performance methods and this is the best way we've found to make it work in both views.

m-m-adams, 2025-01-12:
yeah, that's not possible in the grid. You're welcome to put in a feature request on discord or the discussion tab but this is a feature request. There is no container in rows mode; they are literally just extra clips in the same section and because deluge can only play one clip per instrument at a time the others get muted. We have no feasible way to include this behaviour in the grid because there is only a single pad for each section/instrument combo.

closed 2025-01-12 by m-m-adams

viessg, 2025-01-13:
Hello, 
I  appreciate your feedback very much, however I’m not quite sure that you understood my point. I certainly DIDN’T mean that same instrument or audio loops within single pad in grid view should play at the same time! What I mean – user has ability to select which loop should play from particular pad or in which succession loops should be played when scene (row) is launched. Ok I added this point in discussion section as feature request. 
https://github.com/SynthstromAudible/DelugeFirmware/discussions/3278
Thank you!