SynthstromAudible/DelugeFirmware #3225: CFW 1.2.0 - active mode in Grid View is barely noticable

viessg, 2025-01-06:
### Please describe the problem:

I somewhat struggle with Blue, Green, Red and Pink modes in Grid View - I think it would be a lot easier to distinguish, which mode is active if the pad, which indicates active mode blinked - perhaps that behavior could be set with alternative setting within settings menu. Changes in brightness for Blue and Green modes are so subtle - not easy to read. And there is no any visual feedback of in which recording mode you are Red or Purple.

### What is the expected behavior?

Behavior is probably intended, but it is hard to read - I'd suggest add blinking option for the pad supposed to show active mode in Grid View to improve visibility.

### When did you start noticing the issue?

First time I tried Grid View

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

Especially noticeable this is if you start reducing LED brightness - difference between active and non-active state for Green, Blue. Red and Magenta modes is barely distinguishable

m-m-adams, 2025-01-07:
We're actually aiming to remove the modes entirely. If you try the nightly build and enable "select in green mode" and "default mode:green" you won't need blue mode for anything anymore

m-m-adams, 2025-01-13:
In the nightly I'm pretty sure that copies properly

closed 2025-01-20 by m-m-adams