SynthstromAudible/DelugeFirmware #3237: Horizontal Menus: Sync eats up the screen space

soymonitus, 2025-01-08:
### Please describe the problem:

There needs to be a shorter way of displaying the Sync options (for LFO, and later for Arpeggiator, etc). When the string is longer, like 1-BAR-NOTES, then the Rate parameter is no longer visible in the LFO horizontal menu

### What is the expected behavior?

You can see the three parameters at the same time, Shape, Rate and Sync

### When did you start noticing the issue?

nightly

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

soymonitus, 2025-01-08:
@nikodemus

nikodemus, 2025-01-09:
Rate should not be shown at all when sync is enabled: it does nothing when sync is enabled (unlike with Delay) this is old behaviour. I forget what the popup used to say (not near Deluge right now), but something like "PARAMETER NOT AVAILABLE".

If the Rate is show for an LFO when sync is enabled, that's a bug.

If you would like the Rate to show (and do something) when sync is enabled, that's a feature change.

closed 2025-01-09 by soymonitus