SynthstromAudible/DelugeFirmware #3250: Metronome is always on during count-in and unable to be toggled off

baymud, 2025-01-09:
### Please describe the problem:

The recording count-in (engaged by pressing rec then play) is always accompanied by an audible metronome, regardless of whether the metronome is toggled on or not (shift + tap tempo)

Furthermore, the metronome can not be toggled off during the count-in (via shift + tap tempo) as the 1.2 changelog would suggest (even though that was fixed in #2653)

### What is the expected behavior?

If the metronome is off in general then the count-in should be silent

Also if the metronome is on during count-in you should be able to toggle it off using shift + tap tempo (as per #2653)

### When did you start noticing the issue?

I first noticed it on 2025-01-08 using the 1.2 release

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.2/Chopin

### If possible provide the steps to reproduce the issue and upload additional media:

https://github.com/user-attachments/assets/7d951d1e-a92b-448d-b5cd-2dc1d19120d0

ok-reza, 2025-02-10:
just for some clarification, i believe that the record count-in's metronome has always been separate from the actual metronome-during-playback function. as a result, disabling metronome during count-in technically does work, as the metronome won't sound once actual sequence playback has begun.

however, obviously i agree with you that the two (rec count in, normal metronome) should be paired so that metronome: ON means that metronome sounds will be produced during both counts-in and playback and OFF means no metronome sounds in either case.

baymud, 2025-02-10:
> just for some clarification, i believe that the record count-in's metronome has always been separate from the actual metronome-during-playback function. as a result, disabling metronome during count-in technically does work, as the metronome won't sound once actual sequence playback has begun.

For what it's worth, the reason I created this issue is because I was told that as of 1.2 you should be able to turn the count-in metronome on/off the same way as enabling the regular metronone, but it doesnt' work (there is no way to turn off the count-in metronome, despite what the change log says)

[image]

baymud, 2025-02-10:
> > just for some clarification, i believe that the record count-in's metronome has always been separate from the actual metronome-during-playback function. as a result, disabling metronome during count-in technically does work, as the metronome won't sound once actual sequence playback has begun.
> 
> For what it's worth, the reason I created this issue is because I was told that as of 1.2 you should be able to turn the count-in metronome on/off the same way as enabling the regular metronone (with the result being a silent count-in), but it doesnt' work (there is no way to turn off the count-in metronome). 
> 
> [image]

kastenbalg, 2026-03-10:
No worries. Even the comments right here have different opinions about how this should work. I'm just one opinion.