SynthstromAudible/DelugeFirmware #3255: Audio Export extremely quiet for some clips

Feriixu, 2025-01-10:
### Please describe the problem:

When exporting the clips from my project, one of the tracks is extremely quiet, despite not being quiet in normal playback.

Here is the project along with the samples and my exported clips. The problematic clip is the Warble Bass Pluck.
[project files.zip](https://github.com/user-attachments/files/18371460/project.files.zip)

### What is the expected behavior?

Clips be the same volume after audio export as when playing back.

### When did you start noticing the issue?

just now

### Is there a relevant Pull request?

Don't think so

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

C1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

1. Song View
2. Audio Export Settings: No normalization, Yes export to silence, No Song Fx, Yes Offline Rendering
3. Export Audio by pressing Save + Rec

Feriixu, 2025-01-10:
After further experimentation, I have found another project where this problem exists: 
[problematic project files 2.zip](https://github.com/user-attachments/files/18371738/problematic.project.files.2.zip)
Here the problematic clip is the Melody. It is silent in the export, despite playing in normal playback.

worker-gray, 2025-01-12:
I experienced the same thing on a standard synth track-- no samples, just a subtractive synth layered up with effects

seangoodvibes, 2025-01-12:
Confirming this is a bug with offline rendering being quieter than live rendering

If you try running the audio export with live rendering I think you won't have this issue.

kazemio, 2026-06-13:
Candidate fix: https://github.com/SynthstromAudible/DelugeFirmware/pull/4536.