SynthstromAudible/DelugeFirmware #3256: Song breaks on load with multiple error messages

mcabrerapf, 2025-01-10:
### Please describe the problem:

When trying to load a song multiple errors happen in quick succession

### What is the expected behavior?

Not to crash on song load

### When did you start noticing the issue?

Just now

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_01_10-64ebcdc

### If possible provide the steps to reproduce the issue and upload additional media:

If you download either of this songs and try and load them it should show a bunch of erros https://drive.google.com/drive/folders/15DjBJW6g-FAAhhWMSEMXFjHqC_fU6tlW?usp=sharing

davidvandenbor, 2025-01-19:
Hey! I have the same issue. Chopin 1.2.0 works fine. But EVERY nightly I tried fails with a crash in exactly what the above person describes. Immediate freeze with no option to recover. See screenshot of error screen.

[image]

davidvandenbor, 2025-01-19:
None of the nightly's work. The last nightly I tried was deluge-v1_3_0-nightly+2025_01_12-45c4126.bin. I tried everything, from deleting temp crash song files to force load new songs via startup settings (because I thought it was a particular song) But it crashes no matter which song is designated to load on startup.

davidvandenbor, 2025-01-19:
The Wi-Fi card is also not the problem, because Chopin 1.2.0 works fine. I have the recommended Kingston Canvas Go! Plus card, 64 GB.

davidvandenbor, 2025-01-19:
I (just now) tried to load the latest nightly build 14 (deluge-v1_3_0-nightly+2025_01_19-f8a9c57.bin) to see if a nightly fix could have solved the problem. But alas, the problem still exists. I'll keep using the official 1.2.0 release for the time being.

davidvandenbor, 2025-01-19:
Oh yeah, I have **2 Deluge machines**, both OLED versions!! It crashes on both!! So not a hardware issue either.

m-m-adams, 2025-01-20:
Try tomorrows build, should be fixed

davidvandenbor, 2025-01-23:
Thanks!! I uploaded this night's nightly build. NOW THE FIRMWARE WORKS! Can you shortly mention what the bug or problem was?

m-m-adams, 2025-01-26:
There was a task that was allowed to start even though the SD card was in use, I  added a resource check to the scheduler so that no task can start until it has all the resources it needs to finish their work

closed 2025-01-26 by m-m-adams