SynthstromAudible/DelugeFirmware #3269: Clearing audio clips leaves looper in a weird state

m-m-adams, 2025-01-12:
### Please describe the problem:

When clearing an audio clip with <> + back the looper ends up in a weird state

### What is the expected behavior?

Should be the same as an empty clip

### When did you start noticing the issue?

1.3 looper

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

m-m-adams, 2025-01-20:
Probably related - undo after layering isn't actually switching back to the previous takes audio file either

sapphire-arches, 2025-02-08:
This might be related to #3202 as well.