SynthstromAudible/DelugeFirmware #3273: Error 339 when scrolling kits

x3003, 2025-01-12:
### Please describe the problem:

When press load + kit, then scroll a bit. Sooner or later i get an error 339. Press select will get error m001, then m000, then m998.
Even when the deluge says 'operation resumed' you can't save.

There's no problem with scrolling synths.

### What is the expected behavior?

No error.

### When did you start noticing the issue?

just now

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_01_11-fed3534

### If possible provide the steps to reproduce the issue and upload additional media:

1) power on
2) press load + kit
3) scroll a bit (up and down).

Error + lockup

m-m-adams, 2025-01-20:
This should be fixed as of today (possibly tomorrows nightly build)

x3003, 2025-01-24:
Dunno what you did, but it looks fixed. Unfortunately, the track I was working on when it crashed didn't survive.

closed 2025-01-24 by x3003