SynthstromAudible/DelugeFirmware #3280: Stuck invisible note

DieterVDW, 2025-01-13:
### Please describe the problem:

There is a note playing on a SYNTH track. The note is not visible in the clip, however the tracks view (?) shows that there is SOMETHING there.
Clearing all clip notes using "<> + UNDO" does NOT delete the note.

### What is the expected behavior?

- Playing notes should be visible in the clip view?
-  "<> + UNDO" to clear all notes should clear all notes.

### When did you start noticing the issue?

Just now, installed 1.2 a few days ago.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

### What is the firmware name:

C1.2.0

### If possible provide the steps to reproduce the issue and upload additional media:

Video example: https://photos.app.goo.gl/WWsibZLQRJRBnkt9A

Clues:
- I think I did a few UNDO/REDO commands that affected this note, maybe related to UNDO/REDO logic?
- I was playing around with MIDI follow, turning MIDI follow on and off, and sending notes from an external MIDI controller.

No idea how to reproduce sorry, but if it happens again and I find more clues I'll update.

seangoodvibes, 2025-01-13:
@DieterVDW did you try turning scale off? maybe the note that got recorded is out of scale and is not being shown but it's there

DieterVDW, 2025-01-13:
@seangoodvibes Hmm does  "<> + UNDO" only delete visible notes? I thought that nuked everything in the clip?
I guess that COULD be it, but I can't seem to find how to get into that situation again though.

I tried to simulate this by:
1. Turning off SCALE
2. Setting an out-of-scale note
3. Turning SCALE back on

And a bunch of variations on the above.

But the Deluge seems to do the "correct" thing which seems to be that it switched to USER scale and shows all the notes that are being played. I can't seem to hide notes by changing scales, so it seems at least something is fishy here?

Either way, is it this behaviour expected? Very confusing to have notes playing that are not visible...

seangoodvibes, 2025-01-13:
> @seangoodvibes Hmm does  "<> + UNDO" only delete visible notes? I thought that nuked everything in the clip?
>
> I guess that COULD be it, but I can't seem to find how to get into that situation again [...]

No you're right <> + back should clear all notes

It's def a bug that it's showing a note but it's not visible in the sequencer

Have you tried going into velocity view and seeing if any notes appear there?

Hold audition pad + press velocity and then scroll up and down with the vertical encoder and see if there are any visible notes

DieterVDW, 2025-01-14:
@seangoodvibes Well I forgot to save so I can't reproduce the problem no unfortunately.

Didn't even know about velocity view. But I'll update if I hit it again (and save the song!).

m-m-adams, 2025-10-16:
Can't reproduce so just gonna close this

closed 2025-10-16 by m-m-adams