SynthstromAudible/DelugeFirmware #3287: Attempting to clone synth with defined clip name causes freeze

todd-gochenour, 2025-01-15:
### Please describe the problem:

Cloning synth with clip name causes the Deluge to freeze.  If playing, then it freezes on with a buzz.  Cloning synth with no clip name does not cause issue.

### What is the expected behavior?

To copy clip into new section.  I was testing to see if this copy was with or without the given clip name.

### When did you start noticing the issue?

just now

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_3_0-nightly+2025_01_15-fa3f446.bin

### If possible provide the steps to reproduce the issue and upload additional media:

New Song, SHIFT-NAME, enter any clip name and ENTER to save name.  Song View, click and hold clip row and select empty row below it.  System freezes.  This does not occur on clips with no clip name.

Balistix55, 2025-01-16:
For me also happens with Audio clips

m-m-adams, 2025-01-20:
Is this still happening with todays nightly? We fixed a few string issues in the last couple days

Balistix55, 2025-01-21:
With the new song the problem is gone. With the song created and saved with older nightly deluge still freezes.

todd-gochenour, 2025-01-22:
Defect is still reproducible with nightly deluge-v1_3_0-nightly+2025_01_20-c3c3918.bin.

New Song/FIRST CLIP/SHIFT-NAME/letters-ENTER/SONG/hold FIRST CLIP and press SECOND CLIP

Balistix55, 2025-01-22:
I tried it now folowing your steps todd and you have right, it freezes with synth, midi and cv. Audio clips do not have this issue now

PR #3315 (2025-01-26): reset the string to the  new value
Fix #3287
closed 2025-01-26 by sapphire-arches