SynthstromAudible/DelugeFirmware #3308: Midi thru not working.

Maly-Lemire, 2025-01-23:
### Please describe the problem:

Midi is not sent to midi out when midi thru is enabled.
I have midi follow enabled for one device connected with usb. (Tried disconnecting it)
I am trying to send midi to the din in port from my daw. 
Midi is not sent thru midi out din port or usb.

I tested directly on a device and midi is properly sent from the daw to the device.

### What is the expected behavior?

When midi thru is enabled. I want midi to be sent to out din port and usb from in din port.

### When did you start noticing the issue?

Never tried before. 

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

V1.1.1

### If possible provide the steps to reproduce the issue and upload additional media:

Simply try to send midi to midi in din port with midi thru enabled and test if it's sent to midi din out port and usb.

sapphire-arches, 2025-01-23:
To clarity, is the original MIDI event

sapphire-arches, 2025-01-23:
To clarity, is the original MIDI event coming from USB or the DIN ports? Also, retesting with 1.2/Chopin would be appreciated as we haven't touched lil in some months.

Maly-Lemire, 2025-01-23:
My setup is 

Deluge USB Port -> Roland FP10
Deluge DIN In -> Tascam US-18x08 DIN Out
Deluge DIN Out -> Midi Solution Quadra Thru -> Yamaha DX200
                                                                        -> Dave Smith Tetra
                                                                        -> Korg MS2000 -> Roland JP8000
                                                                        -> Tascam US-18x08 DIN In

- Roland FP10 Is properly learned on the deluge in midi follow mode. (Showing Roland Digital Piano and Channel 1)
- Roland FP10 can receive midi notes from Channel 16.
- I can properly control all midi out instruments from the Roland FP10 by using proper channel.
- I can record on daw from the Roland FP10 by setting a track to listen on the currently followed channel.

(x) When midi thru is enabled. Sending recorded notes from daw to devices does not work. No notes are sent. To neither USB channel 16 to the FP-10 or to DIN out to other devices.
- When connecting directly to a device. (Tested on Yamaha DX200), the notes are properly played.

I will try Chopin next week. Just got a new desktop and ordered an SD Card reader.

x3003, 2025-01-24:
Just tested it w/ the latest dev version. It's working w/o a problem. Maybe you wanna try this version to see if your problem is gone.

Maly-Lemire, 2025-02-01:
An update on this 

- Installed Chopin, changed nothing.
- Tried with Korg MS2000 -> Long MIDI cable -> Deluge MIDI DIN in : MIDI follow and thru working fine.
- Tried with Tascam US-18x08 DIN Out -> Long MIDI cable -> Korg MS2000 and every note is sent properly.
x Tried with Tascam US-18x08 DIN Out -> Long MIDI cable -> Deluge MIDI DIN in : MIDI Follow and thru not working, 
- Tried with Tascam US-18x08 DIN Out -> Long MIDI cable -> Midi solution quadra merge -> Long MIDI cable-> Deluge MIDI DIN in : MIDI follow and thru working fine.

So for now I think the problem can be

- The MIDI signal of the Tascam US-18x08 is to low and need to be boosted by the quadra merge to be detected properly by the deluge. (But it's still strong enough for all my other devices.)
- The Deluge does not handle the input voltage sent by the Tascam US-18x08 and the voltage needs to be lowered by the Midi solution quadra merge power up.

m-m-adams, 2025-02-05:
It sounds like it's not a software bug so closing the issue. You can contact synthstrom and they might know if it's a hardware issue