SynthstromAudible/DelugeFirmware #3309: Not able to export song from arranger

Ponchomouse, 2025-01-23:
### Please describe the problem:

Hi all, great work on 1.2. I have an issue at the minute as unable to export songs from arranger. The export stems works great but hold record and play to export the full song crashes on each song I've tried (about 4 different ones). I get error m123 press select knob.

[image]
[image]

### What is the expected behavior?

It should play the whole way through, crashes after about 1 min of playing. 

### When did you start noticing the issue?

This is the first time I've noticed but not exported for a while. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.2/Chopin

m-m-adams, 2025-01-26:
Does the song play outside of the stem export? Also does the same happen on the nightly? 

for devs - the error is a double free, dunno where it could come from

Ponchomouse, 2025-01-26:
Song plays gone normally and exported via stems fine. Not trued the nightly. Will give it a try and let you know.

Ponchomouse, 2025-01-27:
Just tried the nightly but that would not boot. Reinstalled 1.2 and tried again. It did not crash but created a 0kb file in the relevant folder. Tried a couple of songs and one did the same crash, the other looked to have worked but was a 0kb file created.

Many thanks

sapphire-arches, 2025-02-08:
This might have the same root cause as #3216 as both errors are related to freeing of invalid pointers.

seangoodvibes, 2026-10-05:
Hi @Bitte1Bit and @Ponchomouse 

We merged a couple fixes yesterday and today related to stem export crashes.

As we haven't received further updates from you I will close this issue now.

Please feel free to let us know if the issue persists and then we can revisit at that time.

Regards,

Sean

closed 2026-10-05 by seangoodvibes